DnG-Crafts/ACE-RFID #1: 16bytes or 20bytes ?

xperiments, 2025-03-18:
Hi, comments.....

The first 3 strings can be interpreted as 16 or 20 bytes... what do you think is correct?

Also , last page 31, for me first to bytes seems weight, as parsing from the dumps is across all

Second can be anything... weight unit, remaining weight, etc....

I am adding this logic to xspool.....

Was opening another issue to add sku by you added it XD

Thanks!!

DnG-Crafts, 2025-03-18:
looking at the dumped tag data [SnooCheesecakes1269](https://www.reddit.com/user/SnooCheesecakes1269/) provided the first 3 plaintext blocks all look like they are 5 pages(20 bytes)

```
[ 41:48:50:4C ] Addr. 05 : DATA  sku
[ 50:42:4B:2D ] Addr. 06 : DATA
[ 31:30:32:00 ] Addr. 07 : DATA
[ 00:00:00:00 ] Addr. 08 : DATA
[ 00:00:00:00 ] Addr. 09 : DATA

[ 41:43:00:00 ] Addr. 0A : DATA  brand
[ 00:00:00:00 ] Addr. 0B : DATA
[ 00:00:00:00 ] Addr. 0C : DATA
[ 00:00:00:00 ] Addr. 0D : DATA
[ 00:00:00:00 ] Addr. 0E : DATA

[ 50:4C:41:2B ] Addr. 0F : DATA  type
[ 00:00:00:00 ] Addr. 10 : DATA
[ 00:00:00:00 ] Addr. 11 : DATA
[ 00:00:00:00 ] Addr. 12 : DATA
[ 00:00:00:00 ] Addr. 13 : DATA

```

only reason i thought of 16 bytes was because the read command responds with 16 bytes(4 pages) as shown in section 9.2 of the <a href=https://www.nxp.com/docs/en/data-sheet/MF0ICU2.pdf>MF0ICU2 Datasheet</a>

i figured anycubic just reads the `sku` at address(page) 5 and then all 4 pages are returned in the single read then they skip a page which would be 9 then for the `brand` they read the next 4 by reading page 10 and then skip one page(14) and read 15 for the `type`

it seems to me that is the simplest way to do the reading for those blocks but that is just pure speculation on my part so i decided to just go with 16 bytes because the worst that can happen is the usable string is 4 bytes shorter and none of the data being used on my part is longer than 16.

page 31 i was unsure of but in the ace response there is a "remaining" variable and when you convert the value of E8 03 to dec you get 1000 and considering everything else is metric units i would assume that it is 1000 grams or 1kg which matches the length value of 330.
i was holding off on committing to that assumption until i could see something in the firmware code or logs to confirm it but it does seem likely.

the last 2 bytes of page 31 could be remaining but the only way i could see that being the case is if the ace writes to the tag to track what is on the spool after a print is finished.

i have only had the anycubic printer and ace for 5 days now so i am still testing in between printing things for a review video so i should be able to work more out over time.

xperiments, 2025-03-18:
Topp, great work anyway...

Yes finally I reserved that 4 bytes on each.... of that strings

For the weight... when you find I will update. I don't have any anycubic w me

For the SKU, I will not let users write the one's we know...

To don't make ungry Anycubic....

Anyway I don't get extactly how the maping is done ACE side.... only w material type I suppose, as they can't relly on SKU

Thx!!!!

xperiments, 2025-03-19:
UMMM

Page 4 is not consistent

From the dumps

0x7b006400
0x7b006500
0x7b026540

DnG-Crafts, 2025-03-19:
i observed the same while comparing the dumps which is why i was thinking the last 2 bytes of page 4 were possibly a length value but i could not see a change in the size of data.
the first byte seems to remain the same over all of the dumps which is why i thought it was a identifier byte(magic byte).

i have been using the tags on spools doing test prints and they work well so i think that enough of the tag format has been decoded for it to work.
my next step is to test if the extruder and bed temps in the tags have an effect or if the slicer overrides those values.
if they do have priority over the slicer settings then i will expose them in the code for manual editing.

closed 2025-03-19 by DnG-Crafts

xperiments, 2025-03-19:
Can y confirm works them w ntag213?
I bought some but as I don't have an ACE...

DnG-Crafts, 2025-03-19:
i have a couple of ntag213 on the way to test.
the tags i have here now are ntag215 and they work.

if you look at the dumps they show they are dumped from 213 tags so i assume that the spools have 213 on them but in a few days when my tags arrive in the post i will be able to see.

xperiments, 2025-03-19:
Top, at least ntag215 works.... so mostly they will work aside they check
for size, but 215 size is not same as ultrawire C so mostly 99% will work

Thanks another time