DnG-Crafts/ACE-RFID #7: [Issue] Infos about scanned tags are not set

Dark-Noir, 2025-03-27:
Hey,

if i scan my original Ancubic filament, i only get a "Tag ID", but filament type, color etc. are still default.

AKerambrun, 2025-03-27:
The tag is detected, got an ID but if I hit read, I get an 'invalid tag type' error.
The filament is Anbycubic peach fuzz

evil2k2, 2025-03-27:
I have the same problem with my phone, 
Motorola g84 5G
NFC tools works fine.
I have some spools her if you need a dump from them.

DnG-Crafts, 2025-03-28:
you will always get a tag id on most tags.
the invalid tag type is being thrown when the android api cant communicate with the tag using the [ultralight](https://developer.android.com/reference/android/nfc/tech/MifareUltralight) method.

i will have to source a anycubic rfid spool to test an actual tag because i have never had one and i have been working from dumps other people have provided.
i suspect i may need to add a [ndef](https://developer.android.com/reference/android/nfc/tech/Ndef) method to try and communicate with the tag if ultralight fails.

DnG-Crafts, 2025-03-29:
I pushed an update to the app today that changed the read and write api used to communicate with the tags.
Give it a test and see if it is still failing to communicate with the tags.

AKerambrun, 2025-05-10:
Read works now for me. Still can't write but it has to be a tag type issue. I use aliepress 215 tags which may not be the correct ones