DnG-Crafts/ACE-RFID #10: guide arduino

jhonny811, 2025-03-31:
good morning, I find this app of yours very interesting, but unfortunately I don't have an android phone... but I have an arduino nano and a raspberry pi4, can I kindly ask you for a guide to install on arduino? my idea would be to take the codes from the app you created and write them on the rfid with an iphone.

TheAushan, 2025-04-10:
Same here, I own an iPhone which is capable to write NFC Tags. Maybe a App/ Program/Webpage where you can just download the file to write on the tag?
Worked well with backups of Amiibos (Nintendo)

Buff84, 2025-04-10:
Hello, maybe I found another interesting Page, IT IS Not for iPhone but maybe useable for you.

https://github.com/Molodos/anycubic-nfc-filament

closed 2025-06-28 by DnG-Crafts