DnG-Crafts/ACE-RFID #11: [Issue] Filament not set in Anycubic Slicer Next, when synchronizing

Dark-Noir, 2025-04-02:
Hey,

i created a custom filament in the app and wrote it on the tag:
<details>
https://i.imgur.com/LDxIpuD.png
https://i.imgur.com/ljBTq5E.png
</details>

It shows up like this, the drying temp is only 45°C (should be 50°C for PETG i think):
<details>
https://i.imgur.com/dqfkrOf.jpeg
https://i.imgur.com/HaNSBtJ.jpeg
</details>

In Anycubic Slicer Next it looks like this, when i sync it, it selects the "Anycubic PLA" instead of the correct one.
<details>
https://i.imgur.com/u5rweVO.png
https://i.imgur.com/F5YWZyN.png
https://i.imgur.com/gRebEqh.png
</details>

Is this a bug or is there no way to sync the filament correctly in Anycubic Slicer Next?

boarder876, 2025-04-07:
I encountered the same problem. I wanted to create a third-party filament (e.g. eSun) and save it to the RFID chip. The printer displays the name, but when synchronizing, Anycubic PLA is automatically loaded. The Workbench displays a "?", and when printing starts, eSun is shown on the spool.

But no matter which filament you slice with, the material with the manual RFID cannot be selected.

It seems to me that this is less of an app problem than Anycubic's problem that they don't want to handle third-party RFIDs.

prigodada, 2025-04-16:
Same problem here.
Just can't send gcode to printer from Anycubic Slicer Next but via usb drive all fine

[image]
[image]
[image]
[image]
[image]
[image]
[image]

DnG-Crafts, 2025-04-16:
it looks like they have enforced remote print to only work with anycubic filaments when using rfid tagging

if you tag a spool with one of the system preset filaments it works but if you try with a user preset filament the remote print wont allow it to be sent because it does not see the user preset filament.

DnG-Crafts, 2025-04-16:
anycubic does a check in the background to their servers to verify filaments and it seems they are restricting what they dont recognize.

i can tag with any of the default or "system preset" filaments and it works but if i add a custom filament like "eSun Pla Silk" the slicer will not allow it to be sent because the anycubic workbench function does not recognize the filament.

prigodada, 2025-04-18:
I think the problem is how Anycubic Slicer Next (AS) syncs data with the cloud when it receives information about ACE Pro.

Below is an example of a response from the cloud, where you can see that the "type" field contains all the information, including the supplier, the series, and not just the type. I think this is the problem that makes it impossible to select the filament when sending it to print - there is a discrepancy between the type of "PLA" and "Eryone Pla Matte".

Possible causes of the problem:
1. The information on the labels is written incorrectly.
2. The Anycubic cloud incorrectly stores information from the tag on its servers.
3. ???

```json
"multi_color_box": {

      "slots": [
        {
          "index": 0,
          "sku": "",
          "type": "Eryone PLA Matte",
          "color": [
            255,
            153,
            204
          ],
          "edit_status": 0,
          "status": 5
        },
        {
          "index": 1,
          "sku": "",
          "type": "eSUN PLA Plus",
          "color": [
            239,
            228,
            176
          ],
          "edit_status": 0,
          "status": 5
        },
        {
          "index": 2,
          "sku": "",
          "type": "eSUN PLA Plus",
          "color": [
            38,
            197,
            219
          ],
          "edit_status": 0,
          "status": 5
        },
        {
          "index": 3,
          "sku": "",
          "type": "eSUN PLA Plus",
          "color": [
            0,
            0,
            0
          ],
          "edit_status": 0,
          "status": 5
        }
      ]
    }
```

Dark-Noir, 2025-04-18:
What about LAN mode, without cloud, should be working then or?

prigodada, 2025-04-18:
> What about LAN mode, without cloud, should be working then or?

Does not work :(

The slicer now communicates directly with the printer on the LAN, but now I can't examine the encrypted TLS traffic because I don't know how to get the certificate yet.

DnG-Crafts, 2025-04-18:
the issue is how anycubic recognizes filaments.
because their system is designed only to work with their own filaments when using rfid they are not expecting alternate information in the tags.

if you want to see what the slicer is doing on lan mode you dont need a certificate you just need to use fiddler.

the way the workbench/slicer works is it pulls a list of filaments known for rfid from anycubics server using this api post `/p/p/workbench/api/v3/lan_network/index` with the payload of `{"ver_num":1,"type":"FDM"}` and that will return this [information](https://github.com/DnG-Crafts/ACE-RFID/blob/main/docs/Workbench.md)

initially the work around for custom filament sync was to put the filament name into the type field and that would allow you to sync the correct user preset but it seems that when you go to remote print the job it does not recognize the type because it has the vendor name and serial in the type field so the slicer does not allow you to select it.

i will have a bit of a play with it to see if i can get both sync and remote print to work but i dont think it will work due to the workbench expecting a specific type field and the sync defaulting to anycubic filaments when you put that expected value in the type field.

if that does not work i will end up just disabling the custom filament option and then people can just tag any brand filament as an anycubic filament which should allow it all to work.

DnG-Crafts, 2025-04-18:
yeah it is broken, for now there is no way to get the slicer to sync the filament and remote print using the custom function so i have removed it from the app.

if i manage to find a way around it i will re-enable the function.

closed 2025-04-18 by DnG-Crafts

prigodada, 2025-04-19:
> the way the workbench/slicer works is it pulls a list of filaments known for rfid from anycubics server using this api post `/p/p/workbench/api/v3/lan_network/index` with the payload of [...]

I have seen this request and the response. I didn't understand why there are so many filaments with brand_name = null in the list.

I'll look towards whether we can modify the answer to include custom filaments.

DnG-Crafts, 2025-04-19:
if anycubic actually followed the legal requirements around open source we would have the source for the slicer next and then we could just modify the api request to pull from a local file or something on github but they dont seem to follow open source requirements.