DnG-Crafts/ACE-RFID #13: Older version of ACE RFID

phgagel, 2025-04-25:
I have a small question. Would it be possible to send me the version of your ACE RFID app with the option of typing in the filament name and vendor? You mentioned in the updates that you removed that feature. I am trying to get the ACE working with different filament and suppliers.

DnG-Crafts, 2025-04-26:
when i get a bit of time i will update the app to include the custom filament options again but i will add a message that states the custom settings will not work with remote print.

this will allow the custom option again but also inform the user that the remote print function will not work which should reduce confusion with the option.

closed 2025-04-26 by DnG-Crafts

phgagel, 2025-04-26:
Can you help me with this? I tried to modify your app using android studio but I can't get rid of this error: 
startup failed: only alias(libs.plugins.someAlias) plugin identifiers where `libs` is a valid version catalog
I have no experience programming apps.

DnG-Crafts, 2025-04-26:
the repo was missing the libs toml file, i just added it to the repo

https://github.com/DnG-Crafts/ACE-RFID/blob/main/Android/AceRFID/gradle/libs.versions.toml

phgagel, 2025-04-26:
Thank you for your help and sorry for my questions, but now I get another exception:
An exception occurred applying plugin request [id: 'com.android.application', version: '8.9.1']
> Failed to apply plugin 'com.android.internal.version-check'.
   > Minimum supported Gradle version is 8.11.1. Current version is 8.10. If using the gradle wrapper, try editing the distributionUrl
It says to edit the gradle wrapper version but I can't find it. Is it maybe also missing?

DnG-Crafts, 2025-04-26:
your gradle version is managed by android studio but if you want to edit the distro url it is in the `gradle-wrapper.properties` of a project.

it will contain something like this
```
distributionBase=GRADLE_USER_HOME
distributionPath=wrapper/dists
distributionUrl=https\://services.gradle.org/distributions/gradle-8.11.1-bin.zip
zipStoreBase=GRADLE_USER_HOME
zipStorePath=wrapper/dists

```

phgagel, 2025-04-26:
Ok I added that and now it was able to build the app but I can't load it to the phone:
Unable to delete directory 'C:\Users\phgag\Documents\Privat\AceRFID\app\build\intermediates\data_binding_dependency_artifacts\debug\dataBindingMergeDependencyArtifactsDebug'

DnG-Crafts, 2025-04-26:
cant help with that one sorry, it looks like a permission issue on your computer preventing android studio from deleting a directory.